riscv/riscv-isa-manual #2175: How should the seed register be implemented in the Zkr extension?

shengqilong, 2025-07-31:
I have the following questions about the seed register
（1）The Zkr extension introduces the seed register. How should the OPST field of the seed register be implemented? How do the states transition?
（2）Do CSR write-only and read-modify-write instructions cause changes to the OPST field? If so, how exactly do write-only or read-modify-write operations affect the transition of the OPST field?

ved-rivos, 2025-07-31:
The OPST - output state - is the status of the entropy source. If the entropy source is undergoing a builtin self-test then it would be set to BIST, if the entropy source does not have entropy to provide at the moment it would return WAIT, if it does provide valid entropy then its set to ES16 and the bits 15:0 provide the entropy, and if the entropy source has failed it returns DEAD. During normal operation you may expect to see either WAIT or ES16. 

Typically entropy sources provide a entropy value that needs to be "popped" or "wiped". The the "write" here is intended to do that that popping/flushing/wiping. It is possible that popping an entropy value puts the entropy source into a WAIT state for a while if no additional entropy is available or it may go/stay in ES16 if more entropy is available. It may go to DEAD anytime due to reasons such as detecting a fault during its continual monitoring. It may also go to BIST state at startup or even temporarily when operational to redo the tests.

ved-rivos, 2025-08-01:
Here are a few references to learn about entropy sources: [NIST.SP.800-90B](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-90B.pdf), [RISC-V entropy source interface paper](https://eprint.iacr.org/2020/866.pdf), [OpenTitan ES HWIP](https://opentitan.org/book/hw/ip/entropy_src/index.html). Section 4 of 800-90B discusses health tests, including continuous health tests. The paper on RISC-V entropy source interface has some discussion on implementation strategies.

shengqilong, 2025-08-03:
Below is the impact of CSR instructions on the OPST of the SEED register. Does this seem reasonable?
（1）CSR write-only instructions ignore the impact of write operations on the OPST of the SEED register.
（2）In a CSR read-modify-write instruction, the read operation transitions the OPST from ES16 to WAIT state.

shengqilong, 2025-08-05:
CSR write-only instructions do not read the value of the seed register, so what is the point of using a write operation to wipe entropy?

mickflemm, 2025-08-05:
The seed CSR is a read-only interface to the underlying entropy source, a snapshot of its state. Each snapshot of the source's state we get, that may include a new seed value, must be globally unique and only used once, which is why we always need to update the CSR every time we read it.

However triggering that update is considered a side-effect, and according to chapter 2 of the priv. spec, side effects may only happen on CSR writes, not reads, so we always need to do a write when reading the seed CSR and in the original proposal it was stated that the CSR  "must be accessed with a read-write instruction" (https://github.com/riscv/riscv-crypto/blob/main/doc/scalar/riscv-crypto-scalar-entropy-source.adoc), that the write value is ignored (since we can't influence the state of the entropy source), and that "The purpose of the write is to signal polling and flushing".

That was changed in the privilege spec, which also allows plain writes and not just read-write instructions, where the plain write instruction just triggers the update of the CSR without reading the stored state. I don't know the rationale behind that change, but I guess it can be useful to be able to skip a few snapshots without using them, e.g. if the source is in the WAIT or BIST states and we know that for the next X snapshots we won't get a seed value, or to handle some implementation-specific behavior

Regarding the OPST field and state transitions, as Ved said above, when the underlying source is gathering entropy it's in the WAIT state, when it has enough samples to provide a seed it switches to the ES16 state, if it needs to re-calibrate itself (e.g. due to some change in its environment) it'll temporarily switch to BIST (and back to WAIT or ES16), and if that fails it'll switch to DEAD and remain there. There is a visual representation of the state transitions in the spec btw: https://github.com/riscv/riscv-isa-manual/blob/main/src/images/es_state.svg.

I hope this helps.

gfavor, 2025-08-05:
@mickflemm Thanks for responding.  Regarding the change from read-write instructions to write instructions, did this occur after ratification or before ratification?  The former in general is disallowed except for a very strong good reason.  Did the change happen accidentally because someone was rewriting and not just copy&pasting the spec into the Priv ISA manual?

In short, is this a change that should not have happened (and needs to be reversed), or is there good reason and was acceptable process followed, to make this material change?

mickflemm, 2025-08-06:
I don't know why this was changed and I couldn't find any info on that, there was a question on the mailing list at some point but no answer. I guess we should ask in the Crypto SIG mailing list, or maybe @mjosaarinen knows something about it.

mjosaarinen, 2025-08-06:
> I don't know why this was changed and I couldn't find any info on that, there was a question on the mailing list at some point but no answer. I guess we should ask in the Crypto SIG mailing list, or maybe [@mjosaarinen](https://github.com/mjosaarinen) knows something about it.

First time I hear about it. The final CSR behavior and related privilege flags were iterated with the ARC during the ratification process. They were different in the versions immediately before ratification. Sounds like some kind of editorial error to me. But please fix!

aswaterman, 2025-08-06:
I did some archaeology on this.  What I found was that the ratified spec was deemed ambiguous, in that it very explicitly said "a read-only instruction [...] will raise an illegal-instruction exception" but said no such thing about write-only instructions.  When the question was raised about write-only instructions (in the context of implementing the Spike port), we decided that the spec must have meant that write-only instructions, while perhaps not useful, are not illegal.  This was clarified in https://github.com/riscv-software-src/riscv-isa-sim/issues/1728, which you thumbs-upped, @mjosaarinen :-)

In any case, there was no intention to change the ratified spec; the intention was only to clarify an apparent hole in the description.

gfavor, 2025-08-07:
@mjosaarinen Everyone agrees that read-only accesses are disallowed.  The question is whether write-only accesses (as well as read-write accesses) are allowed (i.e. the current spec), or only read-write acesses (the old spec) are allowed.  Andrew's last post explained where the change from old to current spec came from.

mjosaarinen, 2025-08-07:
> @mjosaarinen Everyone agrees that read-only accesses are disallowed.  The question is whether write-only accesses (as well as read-write accesses) are allowed (i.e. the current spec), or only [...]

I don't see much harm in write-only accesses, as long as it has the same access control as read-write. In "normal" usage it is just causing a flush, and at worst can make the entropy source less available, but a read-write has the same effect. The seed register has some reserved custom bits, so perhaps some vendor would find it useful when setting those. (Entropy sources in secure microcontrollers often require some rudimentary configuration.)

gfavor, 2025-08-07:
Ok then, I think both the original question in this issue and this follow-on spec change issue have all been resolved - and no change to the current spec is needed.  So I'm closing this issue.

closed 2025-08-07 by gfavor

aswaterman, 2025-08-07:
Agreed; thanks, everyone, for helping to get to the bottom of this one.