riscv/riscv-isa-manual #2181: must satp.asid or hgatp.vmid values change be followed by a sfence instructoin?

rock-ifly, 2025-08-04:
As we know, satp.mode = Sv39 -> bare -> Sv39，don't need to use sfence to flush the tlb entries, but satp.mode = Sv39 -> bare -> Sv48 must follow a sfence.

then，
Q1）must satp.asid values change be followed by a sfence instructoin?
It seems that, for e tlb entry with asid, process switch needn't flush the tlb by sfence instruction.

Q2）must hgatp.vmid values change be followed by a hfence instructoin?
It seems that, for e tlb entry with vmid, virtual machine switch needn't flush the tlb by sfence instruction.

Q3, satp.asid=1, satp.mode=Sv39  --> satp.asid=2, satp.mode=Sv48，must flush by sfence since the satp.mode changes, is it right?

Q4, vsatp.mode=Sv39, hgatp.vmid=1  --> vsatp.mode=Sv48, hgatp.vmid=2，must flush by sfence? becase of the vsatp.mode changes,  or the hgatp.vmid changes?

Q5, vsatp.asid=1, hgatp.vmid=1  --> vsatp.asid=2, hgatp.vmid=2，must flush by sfence? becase of the process switch,  or the virtual machine switch?

chara811, 2025-08-04:
For Q4, when performing a TLB lookup for page tables, different virtual machines will not hit each other's page tables because their VMIDs are different. So, for each virtual machine, its system mode has not changed. Is it still necessary to execute HFENCE?

rock-ifly, 2025-08-05:
the spec: hypervisor memory management fence instruction says:

> If hgatp.MODE is changed for a given VMID, an HFENCE.GVMA with rs1=x0(and rs2 set to either x0 or the VMID) must be executed to order subsequent guest translations with the MODE change -- even if the old MODE or the new MODE is bare.

It seems that only the mode change shoule follow a sfence/hfence operation, other fields like vmid or asid changes take effects immediately ,and need not a subsequent sfence/hfence.
Q11: satp.mode changes need  a subsequent sfence? 
       spec/issue says Sv39 --> Sv48 need a sfence, https://github.com/riscv/riscv-isa-manual/issues/2061
Q12: vsatp.mode changes need  a subsequent sfence/hfence.vvma?
       no specs give any hint.
Q13: hgatp.mode changes need  a subsequent sfence/hfence.gvma?
       spec says so.
Q14：If we store asid/vmid, vs-stage/g-stage flag, mode  in every tlb entry,  and compare them in the lkup process, it seems that vmid or asid changes can take effects immediately ,and need not a subsequent sfence/hfence. Am I right?

gfavor, 2025-08-05:
Q11.  Yes.
Q12.  Yes (even though this is happening in HS-mode and vsatp is inactive - but vsatp will become active as satp once the guest is entered).
Q13.  Yes.
Q14.  Yes.  If TLB entries are tagged with ASID/VMID, then there is no need to flush the TLB when ASID/VMID changes occur.

closed 2025-08-07 by gfavor