riscv/riscv-isa-manual #2182: Does different virtual machine configurations require the execution of hfence in different *atp.modes?

chara811, 2025-08-04:
For example, when virtualization is enabled and two-level address translation is performed, with vsatp.mode = sv39, hgatp.mode = sv39, and hgatp.vmid = 1,when a new virtual machine is launched with vsatp.mode = sv48, hgatp.mode = sv48, and hgatp.vmid = 2, is it necessary to execute hfence to flush the TLB entries with vmid = 1?

aswaterman, 2025-08-04:
In general, it is impossible for me to answer the "when do I need to flush the TLB" question, because it depends on the TLB microarchitecture--specifically, which properties are tagged, and whether those tags are complete.  With that said, in a conventional TLB microarchitecture, you probably would not need to flush the TLB in this case.

chara811, 2025-08-04:
So, can it be concluded that, under the current scenario, whether to add Hfence is determined by the microarchitecture?

aswaterman, 2025-08-05:
No, I meant that the action the uarch takes is uarch-dependent.  Whether or not an HFENCE is needed is an ISA property, but what the uarch does with the HFENCE instruction is uarch-dependent.

In this case, I don't think the ISA requires any fence for correct operation.

chara811, 2025-08-05:
Okay, I have a general understanding now. Thank you.

closed 2025-08-05 by gfavor