riscv/riscv-isa-manual #2185: miselect 0x200-0x2FF reserved for the Smctr extension?

noionion, 2025-08-05:
The Smctr extension states:

> The siselect index range 0x200 through 0x2FF is reserved for CTR logical entries 0 through 255. When siselect holds a value in this range, sireg provides access to ctrsource, sireg2 provides access to ctrtarget, and sireg3 provides access to ctrdata. sireg4, sireg5, and sireg6 are read-only 0.
>
> When vsiselect holds a value in 0x200..0x2FF, the vsireg* registers provide access to the same CTR entry register state as the analogous sireg* registers. There is not a separate set of entry registers for V=1.

Does the miselect index range 0x200 throuth 0x2FF also reserved for Smctr extension? If so, when miselect index is in 0x200-0x2FF, should accessing mireg* return read-only 0 or trigger an illegal instruction exception?

noionion, 2025-08-06:
Based on your explanation, it means that when miselect index range 0x200 through 0x2FF, it should be treated as a reserved region, following the hardware model for reserved areas (e.g., accessing the corresponding mireg* will raise an illegal instruction exception, consistent with the *iselect model described in the AIA specification).

> When miselect is a number in a reserved range (currently 0x00–0x2F, 0x40–0x6F, or a number
above 0xFF not designated for custom use), attempts to access mireg will typically raise an illegal
instruction exception.

---

Then I have another question. When the iselect index belongs to an AIA-allocated range (e.g., miselect=0x30 where mireg corresponds to iprio0), should accessing mireg2 raise an exception or read-only 0?

The Smcsrind extension states:

> Attempts to access mireg while miselect holds a number in an allocated and implemented range results in a specific behavior that, for each combination of miselect and miregi, is defined by the extension to which the miselect value is allocated. Ordinarily, each miregi will access register state, access read-only 0 state, or raise an illegal-instruction exception.

Here I believe: When accessing mireg and mireg2 under the same miselect value, they belong to the same address space. Following the Smctr extension's approach:

> When siselect holds a value in this range, sireg provides access to ctrsource, sireg2 provides access to ctrtarget, and sireg3 provides access to ctrdata. sireg4, sireg5, and sireg6 are read-only 0.

Should accessing mireg2 return read-only 0 when miselect=0x30? 

More fundamentally: Should access permissions for mireg/mireg2/mireg3/... be identical under the same miselect value?

_(Aside: The permission design for *ireg accesses became sufficiently complex after the AIA Spec. Now, with Smcsrind becoming a standalone extension, we have additional indirect selector registers – this flexibility equally burdens implementation. Moreover, spike/sail-riscv models lack reference implementations for this.)_

noionion, 2025-08-06:
Based on your reasoning, access permissions for the indirect CSR (mireg/mireg2/mireg3/...) cannot be filtered using a shared miselect value range...

gfavor, 2025-08-06:
If by "shared" you mean that miregi and sregi registers, for the same miselect/siselect values, share the same permissions, then No - since mi* and si* registers exist in separate index value address spaces.

noionion, 2025-08-06:
Oh sorry.  "shared" means that whether accessing different miregi registers uses the same range-checking logic for miselect values, unrelated to si* registers.

For example: Is it possible that with an unchanged miselect value, accessing `mireg` succeeds normally while accessing `mireg2` raises an exception

gfavor, 2025-08-08:
Closing since this issue appears to be resolved.