riscv/riscv-isa-manual #2190: Do FLI.H instructions need to be implemented in the hardware supported by zfa and zvfh extension?

fengxh001, 2025-08-07:
Hi,
    In spec manual 《 "Zfa" Extension for Additional Floating-Point Instructions, Version 1.0》,The description is as follows:
   “ If the Zfh or Zvfh extension is implemented, FLI.H performs the analogous operation, but loads a half-precision floating-point value into register rd.”

    But，in spike and qume , FLI.H only depends on zfh and zfa ; 

  there are conflict between  the spec manual  and the spike.

  My hardware is implemented zfa and zvfh extension，and isn't implemented zfh.  Do FLI .H instructions need to be  implemented？

thanks .

DavidYu360, 2025-08-07:
The spike & qemu should follow the spec, so there is no conflict. :-)
Since your hardware implements Zfa + [Zfh | Zvfh], it should implement FLI.H.

aswaterman, 2025-08-07:
This actually is a Spike bug.  Spike currently only supplies FLI.H if both Zfa and Zfh are implemented.  Instead, it should supply the instruction if `Zfa && (Zfh || Zvfh)`.

I will refile this as a bug against Spike and immediately fix it.  In general, if you suspect a Spike bug, please file an issue on that repo, rather than this one.  But I appreciate the bug report, @fengxh001!

closed 2025-08-07 by aswaterman