riscv/riscv-isa-manual #2203: About not readable behavior

sunhao8886, 2025-08-12:
Hi,

In RISC-V Privilege Specification, Chapter 21.2.6, it says:

[image]

Could you please explain the expected behavior about "not readable"?

Thanks
Hao Sun

aswaterman, 2025-08-12:
Read this section: https://github.com/riscv/riscv-isa-manual/blob/44b827792a6be67e13e6e0688594f3122cf17348/src/hypervisor.adoc#traps

gfavor, 2025-08-12:
I would add on that what makes a CSR "not readable" is that the CSR instruction is being trapped (and hence will not successfully execute and read the CSR).

gfavor, 2025-08-13:
Unless Andrew thinks otherwise, I'm not sure this is practical to do in a consistent way - since this idea occurs everywhere.  I.e. the spec will describe a CSR read and/or write behavior (and may or may not specify some of the qualifiers or conditions on whether the read or write will succeed or not), but there generally are a variety of conditions that will result in the CSR not being readable/writable.

And yes, this example is calling out a condition specific to the counters, but the fact that this AND other conditions can result in the CSR not being readable remains as a general idea across the spec.

aswaterman, 2025-08-13:
I would also prefer to take no action, since the behavior is clearly spelled out elsewhere.

gfavor, 2025-08-17:
Closing since discussion appears to have died down, with the preceding post capturing the resolution.

closed 2025-08-17 by gfavor