riscv/riscv-isa-manual #2207: About the hgatp remaining fields handling when hgatp.mode == bare

sunhao8886, 2025-08-14:
Hi,

In the RISC-V specification, about hgatp register, it says:

[image]

About the two "UNSPECIFIED", I have several questions:
(1) Is that possible that the specification can provide specific behavior? What is the reason why specification indicates "UNSPECIFIED"?

(2) If it needs to be UNSPECIFIED, I would like to make sure:
_When hgatp.mode == bare, renaming fields of hgatp are not zero, hardware can have the following behaviors?
A. report trap
B. not perform G-stage translation and protection
C.  perform G-stage translation and protection_

Thanks
Hao Sun

gfavor, 2025-08-14:
First note that the Unpriv ISA Manual provides the definition of "UNSPECIFIED".

1) Specifying more specific behavior would be a backward incompatible change to a ratified spec.  So generally not possible.  Th evalue of specifying behavior as UNSPECIFIED is providing flexibility to an implementation to do what it considers simplest and/or best.

2) In the text in question, the UNSPECIFIED behavior is specific to what value the other fields can assume, and what is the resulting translation/protection behavior.  So yes to B and C.  But there is no mention of trapping behavior becoming UNSPECIFIED - so no to A (especially since there is no currently specified trapping behavior associated with field values within this CSR).

sunhao8886, 2025-08-15:
Thanks so much. It is clear

closed 2025-08-15 by gfavor