riscv/riscv-isa-manual #2208: Can G-stage support sparse modes?

sunhao8886, 2025-08-14:
Hi,

In Specification, it says:

[image]

Could you please help confirm whether the following hardware does not violate RISC-V specification?
_Support: Sv57x4, Sv39x4
Not Support: Sv48x4_

Additionally, about writing unsupported value of hgatp.mode, it says:

Could you please help explain why it is different from satp?

Thanks
Hao Sun

aswaterman, 2025-08-14:
I do not believe that configuration is intended to be valid.  Sv57 requires Sv48, and Sv48 requires Sv39.  I assume the intent was that the hypervisor ISA provides Sv57x4 when Sv57 is provided, Sv48x4 when Sv48 is provided, etc., in which case Sv57x4 implies the existence of Sv48x4.  However, I cannot find the statement in the hypervisor spec that imposes this requirement.  Maybe @jhauser-us knows.

aswaterman, 2025-08-14:
Okay, I found the answer.  The H extension itself does not impose this requirement, so technically Sv57x4 without Sv48x4 conforms to the H extension spec.  However, the profiles with hypervisor support include the Shgatpa extension, which does impose the requirement.  See https://github.com/riscv/riscv-profiles/blob/5879c13c924ec5636995c5883f40337e83f6049a/src/profiles.adoc#623-rva22s64-optional-extensions -- note that H is not an option; only Sha (which includes Shgatpa) is an option.

So, to conform to the profiles, you do need to implement Sv48x4 and Sv39x4 if you implement Sv57x4.

aswaterman, 2025-08-15:
Yeah, your verification concern is entirely valid, but this is the design intent.  The reason why is that the supervisor ISA says that if e.g. Sv57 is supported, then Sv48 must be supported, too.  If the setting the hypervisor to Sv57x4 disabled the guest OS's ability to select Sv48, then it would break that contract.

sunhao8886, 2025-08-15:
> Yeah, your verification concern is entirely valid, but this is the design intent. The reason why is that the supervisor ISA says that if e.g. Sv57 is supported, then Sv48 must be supported, too. If [...]

Thanks so much for your explanation.
So I would like to confirm whether the following combination usage is also needed to support?
**VS-Stage: Sv57
G-Stage: Sv48x4**

I am not sure whether the scenario above will not be used by software at all. If so, this combination brings cost to hardware design and verification.

aswaterman, 2025-08-15:
Yes, that's what is required by the Shgatpa extension, hence the RVA/RVB profiles.

sunhao8886, 2025-08-15:
Okay, thanks very much.