riscv/riscv-isa-manual #2210: some question about access control by the mstateen0/hstateen0

shengqilong, 2025-08-15:
```
A virtual-instruction exception is raised for attempts from VS-mode or VU-mode to directly access vsiselect or vsireg*, 
or attempts from VU-mode to access siselect or sireg*.
```

```
When mstateen0[60]=0, an attempt to access one of these CSRs from a privilege mode less privileged than M-mode 
results in an illegal-instruction exception.
```

```
However, regardless of any other mstateen bits, if mstateen0[60] = 1, a virtual-instruction exception is raised as 
described in the previous section for all attempts from VS-mode or VU-mode to directly access vsiselect or vsireg*, 
and for all attempts from VU-mode to access siselect or sireg*.
```

If the smstateen/sscsrind extensions are implemented, here's my understanding - please confirm if it's correct.
(1) when mstateen[60]=0, access siselect/sireg* from vs mode raise illegal instruction exception
(2) when mstateen[60]=0, access vsiselect/vsireg* from vs mode raise virtual instruction exception
(3) when mstateen[60]=0, access siselect/sireg* from vu mode raise virtual instruction exception
(4) when mstateen[60]=0, access vsiselect/vsireg* from vu mode raise virtual instruction exception

shengqilong, 2025-08-17:
```
The AIA bit controls access to AIA CSRs siph, sieh, stopi, hidelegh, hvien/hvienh, hviph, hvictl, hviprio1/hviprio1h, 
hviprio2/hviprio2h, vsiph, vsieh, and vstopi, as well as to the supervisor-level interrupt priorities accessed 
through siselect + sireg (the iprio array of [Section 5.4.1]).
```

if smaia/ssaia extensions are also implemented, when mstateen[59]=0 and siselect=0x30(iprio0), what type of exception should be raised when accessing sireg from VS/VU mode?

gfavor, 2025-09-03:
The third excerpt indicates that the AIA bit controls state not controlled by the CSRIND bit, and the second excerpt enumerates the CSRs that are controlled by the AIA bit.  This list does not include general access to the *select and *reg CSRs - which are controlled by the CSRIND bit.

So, if CSRIND generally prevents access to the *select and *reg CSRs, then of course all CSRs accessed indirectly (including those controlled by AIA) cannot be accessed.  And conversely, if CSRIND allows access to the *select and *reg CSRs, then AIA may or may not allow access to its specific subset of indirect CSRs.

closed 2025-09-03 by gfavor