riscv/riscv-isa-manual #2222: Clarification on preserve program order rule 5 (A has an acquire annotation)

jrahmeh, 2025-08-21:
Given an lr.aq instruction A, and a load instruction B that follows A in program order, can B be observed to perform before A in global memory order if A and B have different data addresses?  Or will this violate PPO rule 5?

This is almost identical to the scenario described in section A.3.2 (load value axiom) of the unprivileged spec except that we have lr.aq instead of a fence.

In the following example (similar to that of section A.3.2):

```
hart 0                                    hart 1
        li           t1, 1                          li           t1, 1
(a)   sw        t1, 0(s0)           (e)   św        t1, 0(s1)
(b)   lr.w.aq  a0, 0(s0)          (f)   lw         a2, 0(s1)
(c)   lw         a1, 0(s1)            (g)  fence   r, r
                                               (h)  lw         a3, 0(s0)
```

Is this final outcome allowed:  a0=1, a1=0, a2=1, a3=0 

Here's one way it could happen (similar to section A.3.2):

(a) executes and goes to store buffer
(b) executes and forwards from (a):  a0=1
(c) executes since (b) has completed, it loads 0 from memory: a1=0
(e) executes and goes to store buffer
(f)  executes and forwards from (e): a2=1
(g) executes since all previous loads have completed
(h) executes and reads 0 from memory: a3=0
(a) drains
(b) drains

gfavor, 2025-08-22:
>
> Given an lr.aq instruction A, and a load instruction B that follows A in
> program order, can B be observed to perform before A in global memory order
> if A and B have different data addresses? Or will this violate PPO rule 5?
The Atomics spec says the following regarding when the aq bit is set for an
atomic operation: "no following memory operations on this RISC-V hart can
be observed to take place before the acquire memory operation".  There is
no mention of this being on a per-address basis (and that is
the correct intention of the architecture).

So I believe this answers your question.

Greg

gfavor, 2025-08-26:
Along the lines of my last post, the ".aq" only orders following loads/stores with respect to the lr.w.aq; it does not order following loads/stores with respect to loads/stores - such as (a) - that precede the lr.w.aq.  It just requires what it says - that (c) must appear after (b) in the global order.

This matches PPO rule 5 which says "An acquire operation, as would be used at the start of a critical section, requires all memory operations following the acquire in program order to also follow the acquire in the global memory order"

gfavor, 2025-08-27:
Closing since this issue appears to be resolved.

closed 2025-08-27 by gfavor