riscv/riscv-isa-manual #2237: Sstc: why QEMU and Linux turn off timer interrupts when all ones in xtimecmp CSRs

oolegoon, 2025-08-27:
Hi! 
Could someone please clarify why QEMU (https://lists.gnu.org/archive/html/qemu-devel/2024-10/msg00242.html) and Linux (https://lore.kernel.org/all/20230710131902.1459180-2-apatel@ventanamicro.com/) rely on timer interrputs turning off when write all ones in xtimecmp registers?
I didn't find any statements in spec about this behavior.

ved-rivos, 2025-08-27:
On Wed, Aug 27, 2025 at 07:46:23AM -0700, oolegoon wrote:
>oolegoon created an issue (riscv/riscv-isa-manual#2237)
>
>Hi!
> Could someone please clarify why QEMU (https://lists.gnu.org/archive/html/qemu-devel/2024-10/msg00242.html) and Linux ***@***.***/) rely on timer interrputs turning off when write all ones in xtimecmp registers?
>I didn't find any statements in spec about this behavior.

The qemu-devel thread has the explanation:

   "While the spec doesn't state it, setting timecmp to
    UINT64_MAX is another way to stop a timer, as it's
    considered setting the next timer event to occur at
    infinity. And, even if the time CSR does eventually
    reach UINT64_MAX, the very next tick will bring it
    back to zero, once again less than timecmp."

If the time is close to overflow then it is possible that 
momentarily the timer interrupt pending will get asserted.

This follows from Sstc:
   A timer interrupt becomes pending whenever time contains
   a value greater than or equal to timecmp, treating the
   values as unsigned integers. If the result of this
   comparison changes then this will reflect eventually in
   the timer interrupt pending signal.

regards
ved

oolegoon, 2025-08-27:
If an interrupt is set for 1 clock cycle, and the hardware takes the interrupt into processing entering trap handler, is this a uarch bug?

Or it is ok and SW should take into account that this is possible according to statements from spec below:
```
If the result of the comparison between mtime and mtimecmp changes, it is guaranteed to be reflected in
MTIP eventually, but not necessarily immediately.

A spurious timer interrupt might occur if an interrupt handler increments mtimecmp then
immediately returns, because MTIP might not yet have fallen in the interim. All software
should be written to assume this event is possible, but most software should assume this event
is extremely unlikely. It is almost always more performant to incur an occasional spurious
timer interrupt than to poll MTIP until it falls.
```

ved-rivos, 2025-08-28:
I am not familiar with this qemu code. You may want to discuss this on the qemu developers list. Instruction set simulators like qemu dont intend to be cycle accurate and so can take certain liberties such as not making the assertion of STIP observable for the one clock cycle where time rolls over from all 1s to all 0.

aswaterman, 2025-08-28:
> and so can take certain liberties

This one is a little worrisome, though, since it could inadvertently hide software bugs (which could only manifest if `time` was initialized to a very large value).

Anyway, the idea is that setting `timecmp` to 2^64-1 works in practice if `time` was initialized to 0, because we'll all be dead before `time` can increment 2^64-1 times.  But, as Ved said, that doesn't mean that the value 2^64-1 is a sentinel value that suppresses the interrupt from firing.  If I were the QEMU devs, I'd remove this special case.

ved-rivos, 2025-08-28:
The idea with 2^64-1 is that at the rollover point when time == 2^64-1, the STIP asserts and then the very next cycle the time becomes 0 and STIP deasserts. So really one can observe it for exactly one cycle.

pdonahue-ventana, 2025-08-28:
> So really one can observe it for exactly one cycle.

It depends on the definition of cycle.  It's one cycle of the mtime clock which in real systems would usually be much slower than the hart clock.  (I'm not sure how qemu handles this type of thing.)

aswaterman, 2025-08-28:
As a litmus test, the following code will very likely (but not necessarily) generate a timer interrupt on a real system:

```
mtimecmp := 2^64-1
while (1)
  mtime := 2^64-1
```

But it will never do so on QEMU.  This is benign in some sense, but is problematic in another (it won't catch software bugs related to mtime overflows).

And since QEMU is going out of its way to make this unusual behavior a special case, it makes even less sense to me.

ved-rivos, 2025-08-28:
Agree. I don't see why Qemu makes a special case for all 1s timecmp value. The behavior around time rollover is well defined.

gfavor, 2025-09-02:
Closing since this is an ISA manual issue that has been resolved as far as specs are concerned.

@aswaterman @ved-rivos Can one of you make the change to QEMU or arrange for someone else to do that?  And does that need to be tracked somewhere else (or re-open this issue here if this is the place)?

closed 2025-09-02 by gfavor

aswaterman, 2025-09-03:
@frankchang0125 would you be able to handle this one?

gfavor, 2025-09-10:
On Wed, Sep 10, 2025 at 10:32 AM oolegoon ***@***.***> wrote:

> *oolegoon* left a comment (riscv/riscv-isa-manual#2237)
> <https://github.com/riscv/riscv-isa-manual/issues/2237#issuecomment-3275884201>
>
> Thanks for clarification and attention to my question!
>
> I have another one question, could you please clarify it.
> mtimer and mtimecmp are memory-mapped registers.
> stimecmp and vstimecmp are CSRs.
> As I understand, this is made because mtimer is general for all harts in
> system. Because of this it has some transport level for timer value
> delivery for RDTIME pseudoinstruction.
> mtimecmp is located near mtimer because it is MM register also.
> So it is necassary to deliver mtimer value and MTI for MTIP.
>
Generally one would not "deliver" mtime and mtimecmp values to a hart to
then compare and produce MTIP.  Instead the compare is down wherever mtime
and mtimecmp are located, and just a timer interrupt request signal is sent
to a hart (to become MTIP).

But, to be clear, mtime is separately "delivered" to a hart for the sake of
becoming 'time' inside of the hart.  And for time then being used in
compares with 'stimecmp'/etc.

> s/vstimecmp CSRs is located inside hart, and STI uses delivered value.
> Latency of transport level is covered by spec statement:
> If the result of the comparison between mtime and mtimecmp changes, it is
> guaranteed to be reflected in MTIP eventually, but not necessarily
> immediately.
> In case when mtimer value is delivered before MTI and mtimecmp and
> s/vstimecmp contain the same values there can be situation when STI is
> taken instead of MTI because of transport level. It this situation
> consistent with spec or this is bug in transport level?
>
There can still be a race between the timer interrupt request signal
getting to a hart and becoming MTIP, and mtime values getting to a hart,
being compared with stimecmp, and producing STIP.  There is no ordering
guarantee between these two sides of this race.

Greg

Message ID: ***@***.***>
>