riscv/riscv-isa-manual #2240: [Smrnmi] In RNMI handler, Does the hardware need to intercept the execution of the mret instruction?

wuliang08, 2025-08-28:
[image]

    
 When software is in RNMI handler, if software send a MRET instruction to return from RNMI handler, the pc will redirect to mepc, not mnepc. 
Should hardware implement need to raise an illegal-instruction exception, or just ignore the excution of MRET in RNMI handler?

aswaterman, 2025-08-28:
No, the behavior is well defined.  Normally, executing MRET in this situation is a software bug, but it is technically a legal thing to do.

gfavor, 2025-09-02:
Closing since it's been a week and this issue appears to be resolved.