riscv/riscv-isa-manual #2245: [CMO] can we implement xenvcfg.CBIE without invalidate (2b01) support?

Steven-Li-Xiaogang, 2025-08-29:
https://github.com/riscv/riscv-isa-manual/issues/2239

> When executing a cbo.inval instruction, an implementation may instead perform a flush
> operation, since the result of that operation is indistinguishable from the sequence of
> performing a write transfer to memory just before performing an invalidate operation.

Spec also said that Software should discovery ‘CBIE support at each privilege level’，does it means that we are able to implement part of the setting?

Thanks,
Steven

pdonahue-ventana, 2025-08-29:
Yes. It's WARL so you can implement whatever subset of modes you like (unless you claim to support some other extension that says that you must support some particular mode).

Steven-Li-Xiaogang, 2025-08-29:
@pdonahue-ventana thank you for clarification