riscv/riscv-isa-manual #2300: the function of the C bit in mstateen0/hstateen0/sstateen0

shengqilong, 2025-09-28:
```
The C bit controls access to any and all custom state. The C bit of these registers is not custom state itself; 
it is a standard field of a standard CSR, either mstateen0, hstateen0, or sstateen0.
```
Does the C bit in mstateen0/hstateen0/sstateen0 control access to custom registers?

If I define a custom S-mode register scpuctl (0x9C0), should accessing scpuctl in HS-mode trigger an illegal instruction exception when mstateen0.C = 0?

Steven-Li-Xiaogang, 2025-09-29:
@gfavor 

> Each standard-defined bit of a stateen CSR is WARL and may be read-only zero or one
> 
> The requirements that non-standard extensions must meet to be conforming are not
> relaxed due solely to changes in the value of this bit. In particular, if software sets
> this bit but does not execute any custom instructions or access any custom state, the
> software must continue to execute as specified by all relevant RISC-V standards, or
> the hardware is not standard-conforming.

I the custom registers doesn't take effect on all architecture behaviors (including executing instructions and access CSRs), would it be necessary to implement C bit?

gfavor, 2025-09-30:
Closing since this issue appears to be resolved.

closed 2025-09-30 by gfavor