riscv/riscv-isa-manual #2302: Zvkt extension specification: Shift instructions latency depends on shift amount provided by RS1, hence RS1 value should be exempt from DIEL

empnik, 2025-09-28:
In Zvkt extenstion specification, chapter 2.15. it is specified:
"_In some cases --- which are explicitly specified in the lists below --- operands that are used as control
rather than data are exempt from DIEL._" 

But further on, when shift instructions are listed, on paragraph 2.15.13:
• vnsr[al].w[ivx]
• vsll.v[ivx]
• vsr[al].v[ivx]
There is no explicit mention of RS1 value being used for control of these instructions, as it should be exempt from DIEL.
I think this is a mistake or oversight.

As a reference, RS1 is correctly mentioned and exempt as control operand for **vslide[up|down].vx**

nibrunieAtSi5, 2025-09-28:
Thank you for moving your questions to **riscv-isa-manual** @empnik 

(copying part of my response on the original issue https://github.com/riscv/riscv-crypto/issues/401)

The issue is raised about this section of the spec https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc#zvkt---vector-data-independent-execution-latency.

First of all, the immediate operand is considered part of the opcode and not a data operand for Z(v)kt and so the immediate in .vi is generally not covered by the DIEL mandate (it is openly visible in the binary code anyway).

> As a reference, RS1 is correctly mentioned and exempt as control operand for vslide[up|down].vx

I think we made the distinction because the `rs1` in `vslide[up|down].vx` can be seen as a control parameter for a permutation-like operation and that it may be difficult and not that useful to force DIEL here. For example some implementations may want to implement differently when the value of `rs1` is a multiple of `VLEN / SEW` for example (which corresponds to sliding full vector registers).

The shift amount is not seen as a control operand.

The scalar Zkt (https://github.com/riscv/riscv-isa-manual/blob/73b7dc8bfc0053bda1fc7a26585be96fb9afc816/src/scalar-crypto.adoc#zkt-listings) does not make the distinction between the shift data operand and the shift amount. 

Also note, that Zvkt extends the DIEL mandate to vector rotations from Zvbb (and widening shifts). Rotation and shift amounts have very similar behaviors. So the whole family should be considered similarly when it comes to DIEL mandate on shift/rotation amounts.

gfavor, 2025-09-28:
Your concern that rs1 for shift instructions was incorrectly not specified as exempt from DIEL should be raised with the crypto group on their email list (tech-crypto-ext).  They are the ones to either explain why rs1 is not exempt, or to agree that it should have been.  (If the latter case, they would also be the ones to explore how to go about addressing that.)

So, given that this concern needs to be raised elsewhere and, if necessary, resolved by parties elsewhere, I'm closing this issue.  Any change (if allowed) to the current ratified spec would need to be handled through a process that eventually would culminate in a separate GH issue and spec PR explained and justified as a result of that process.

But the first step is to talk with the crypto people as to what is the correct Zvkt intent.

closed 2025-09-28 by gfavor